limpkin/cdm324_fft #2: Please share some further infos about this project

will-code-for-pizza, 2023-10-02:
Hi Limpkin.fr

Please provide us beginners with some more infos about the CMD324 FFT:

* Fritzing.org breadboard cabeling example
* Detailed information about the capabilities of onboard connectors shown in [THIS PICTURE](https://www.limpkin.fr/public/cdm324_v2/exp_pinout.png)
* An example Arduino sketch file, that shows, how to print out measured speed on serial console with high accuracy

Many thanks so far.
Have a nice day.

limpkin, 2023-10-03:
Hello,
I'll try to get to it when I have the time.

will-code-for-pizza, 2023-10-03:
Sorry, please don't get me wrong, but you sell your excellent product online without any detailed documentation. It would be nice for every of your customers to follow your developement cabling and Arduino code (which should exist due to your journey of building this backpack V2)

Please upload here and we will find the time to search into it for the needed information for our ongoing projects. 

Many thanks.

limpkin, 2023-10-03:
Just to make sure I follow you, are you saying that https://www.limpkin.fr/index.php?post/2022/03/31/CDM324-Doppler-Motion-Sensor-Backpack%2C-now-with-FFTs%21 isn't detailed documentation?

I'm not sure to follow your reasoning: this CDM324 backpack is a **redesign** of its previous version. I therefore don't understand why it should carry its previous version legacy.

In any case, I've updated the readme at https://github.com/limpkin/cdm324_fft to provide basic instructions on how to get the speed through UART. That should hopefully unblock you

will-code-for-pizza, 2023-10-09:
From a developer's perspective, things often seem simple because you had to go through them yourself during the development process, solve the problems and then put the problem aside.

As a customer, you are faced with the challenge of receiving a foreign device with which you have had no contact whatsoever.

I'm trying to illustrate my problem with your documentation (which from your point of view certainly seems extensive and complete) using an example.
The addition to your readme helped me a lot.
Thanks for that!

`
UART connection

To connect the device to an external platform through UART, please use the MCU UART RX/TX pins shown above.
The baud rate is 1Mbit/s, speed can be queried by sending the 'k' character for km/h or the 'm' character for mph.
`

Ok, I understand to connect MCU UART_RX to Arduino Pin 2 (RX) and MCU UART_TX to Arduino Pin 3 (TX).
I have to initialize the UART connection to 1MBit/s with 

mySerial.begin(1000000);

and send a "k" (ASCII char 107) to Arduino TX pin.

limpkin, 2023-10-10:
Hello,

- the DOPPLER_OUT isn't part of the UART connection, which is why it is not documented in the UART connection paragraph
- as you're a developer, did you make sure that the Arduino board you're using can actually handle a 1Mbit/s baud rate?

limpkin, 2023-10-10:
> I am using the newest Arduino UNO R4 Wifi, but you are right... It could be that the baud rate is too high for this device.
Could you please provide me with the hardware YOU are using for this ?

I'm not using any hardware, as I've only been using the expansion board

> Would it be possible to use DOPPLER_OUT to calculate the speed with the arduino ?

yes, but that would involve basically doing everything that the on board MCU is already doing.

will-code-for-pizza, 2023-10-10:
> yes, but that would involve basically doing everything that the on board MCU is already doing.

Except the amplification of the signal and accordingly the improved sensitivity of the CDM324, right ?

What would the wiring to an Arduino look like in this case?
Do the signals arrive at the Arduino analog or digital?

Is it also possible to operate the radar module with a different baud rate? (e.g. with 0.5 MBit/s)

Regards.

limpkin, 2023-10-10:
> Except the amplification of the signal and accordingly the improved sensitivity of the CDM324, right ?

the MCU isn't doing any amplification, it's just running an FFT. DOPPLER_OUT is the signal going into the MCU ADC.
You'd therefore have to wire that signal to your board ADC.

As you're a developer, I'm sure you'll be able to recompile these source files to get a new baud rate.

will-code-for-pizza, 2023-10-10:
Hi...

> yes, but that would involve basically doing everything that the on board MCU is already doing.

But I would at least benefit from the signal amplification. Right?
What would the wiring to the Arduino look like in this case?
Are the signals then digital or analog?

Is it also possible to operate the radar module with a different baud rate? (e.g. with 0.5 MBit/s)

Best!

limpkin, 2023-10-10:
> But I would at least benefit from the signal amplification. Right?

correct.

> What would the wiring to the Arduino look like in this case?

connect that signal to your ADC input

> Are the signals then digital or analog?

analog, as extensively described in the website article

> Is it also possible to operate the radar module with a different baud rate? (e.g. with 0.5 MBit/s)

Yes, see my previous answer on that topic.

will-code-for-pizza, 2023-10-10:
Sorry for the double post, my browser (Brave) goes crazy...  :-(   Thanks for the quick response.

will-code-for-pizza, 2023-10-10:
> As you're a developer, I'm sure you'll be able to recompile these source files to get a new baud rate.

No, I am not a developer, I'm a normal person who wants to evaluate a radar signal with an Arduino and display it on a website.
I will try my best.

Maybe that explains all my questions. :-(

will-code-for-pizza, 2023-10-12:
Hi Mathieu, thanks for the hint/link.

I have decided to start from the very beginning and connected your CDM324 to my Raspberry Pi, which can handle up to 3 MBit/s UART.

Cabling:

```
CMD324                  <-->               < -->                 Raspberry Pi
=============================================================================
5V_IN                   external 5V HW-131 5V OUT
GND                     external 5V HW-131 GND
MCU UART_RX                                                      GPIO 15
MCU UART_TX                                                      GPIO 14
```
Reference:
https://www.limpkin.fr/public/cdm324_v2/exp_pinout.png
https://www.elektronik-kompendium.de/sites/raspberry-pi/fotos/raspberry-pi-15b.jpg

(UART connection was tested before on Raspberry Pi (TX/RX direct connection and sending/receiving Strings))

In first terminal:
```
# cat /dev/ttyAMA0
```

In second terminal:
```
Set Baud rate:

# stty -F /dev/ttyAMA0 1000000

Check port setup:

# stty -F /dev/ttyAMA0

Send 'k' to serial:

# echo "k" > /dev/ttyAMA0

or with line feed

# echo -e "k\r\n" > /dev/ttyAMA0

```

In first terminal only multiple empty lines or some lines with 'k' appear.
No string with any speed value.

Best.

limpkin, 2023-10-12:
Similarly to our discussion about Arduino's, did you make sure the Rpi can actually handle 1mbit/s? is your wiring short enough?
[cdm324.zip](https://github.com/limpkin/cdm324_fft/files/12878987/cdm324.zip)
attached here is the firmware for a 57600 baud rate.
To flash it, you could use the stm32loader Python module, **only if you also purchased the expansion board with your sensor** https://pypi.org/project/stm32loader/
Command is: stm32loader -b 115200 -p  com_port -e -w -v -s -f F3 file.bin

otherwise, you'll need to do the reset/boot sequencing yourself.

will-code-for-pizza, 2023-10-12:
Hi Mathieu,

as I wrote above, Raspi can handle up to 3 MBit/s (according to documentation, I haven't found any project, that uses so high baud rates...)

My wiring is very short. I am using 20cm breadboard cables with a direct connection from the module to Raspi.

I have purchased the expansion board, one sensor with LED and one without.

Let's try the 57600 baud firmware. Many thanks for that!

will-code-for-pizza, 2023-10-12:
By the way: 
Whats the correct command to store actual firmware from module to file ?

```
stm32loader -b 115200 -p /dev/ttyUSB0 -r -s -f F3 -a 0x8000000 -l 0x00063 cmd324_1mbit-baudrate.bin
```
generates a much bigger file. I will hope, it is restorable...

limpkin, 2023-10-12:
I actually don't know, but I'm guessing it's documented...

will-code-for-pizza, 2023-10-12:
Result with new 57600 firmware:

Same as here
https://github.com/limpkin/cdm324_fft/issues/2#issuecomment-1759123796

but after sending a few "k's", communication hangs up... 

```
root@radar:~# echo "k" > /dev/ttyAMA0
root@radar:~# echo "k" > /dev/ttyAMA0
root@radar:~# echo "k" > /dev/ttyAMA0
root@radar:~# echo "k" > /dev/ttyAMA0
root@radar:~# echo "k" > /dev/ttyAMA0
^[[A
```
:-(

limpkin, 2023-10-12:
I'm guessing something is wrong with your setup... you're the third person flashing that custom firmware into this board, the two others managed to make it work. 
if you're using the expansion board, make sure that you're correctly setting the CTS/RTS signals: https://github.com/limpkin/cdm324_fft/blob/main/scripts/cdm324_device.py#L29

will-code-for-pizza, 2023-10-12:
[image]
[image]
[image]
[image]

will-code-for-pizza, 2023-10-13:
@limpkin Good morning :-)

Could it be that I have to pass the "k" to the module as RAW data? Just an idea...

https://klafyvel.me/blog/articles/fft-arduino/#dont_bother_with_communication_protocols_over_serial

limpkin, 2023-10-15:
it depends on your library... at the end of the day you need to send 'k', which is the number 107 in ascii :)

will-code-for-pizza, 2023-10-15:
Hi Mathieu, as you can see in all my code examples from Arduino to Raspberry, from a sketch to simple Bash commands, sending plain 'k' string or ASCII 127 to your module, with baud rates from 57600 to 1 MBit/s .... nothing happens. May I ask you, if you have ever tried this approach by yourself successfully?   Have a nice day. Regards.

will-code-for-pizza, 2023-10-16:
Update with cdm324 FFT 57600 baud:

UART testing sketch:

```
#include <Arduino.h>
#include <SoftwareSerial.h>

#define ARRAY_SIZE 10 // Size of array to store data
char receivedData[ARRAY_SIZE]; // Array to store data into
int dataCounter = 0; // Counter for received data

void setup() {
  // Init serial console baudrate
  Serial.begin(57600);
  
  // Init UART console baudrate
  Serial1.begin(57600);
  SoftwareSerial Serial1(0, 1); // RX, TX
}

void loop() {
  // Send string 'k' via UART
  Serial.println("Sending data...");
  Serial1.println("k");

  // Waiting for UART data
  if (Serial1.available()) {
    // Check if there is space left in array to store data
    while (dataCounter < ARRAY_SIZE) {
      Serial.println("Space left in array");
      char receivedChar = Serial1.read();
      receivedData[dataCounter] = receivedChar;
      dataCounter++;
    }

    int size = sizeof(receivedData);
    Serial.print("Array full. "); 
    Serial.print("Size:");
    Serial.println(size);

    for (int i = 0; i < ARRAY_SIZE; i++) {
      Serial.print(receivedData[i]);
    }
    Serial.println(); // New line
    // Reset counter
    dataCounter = 0;
  }
  delay(5000);
}
```

Result:

```
Sending data...
Sending data...
Sending data...
Sending data...
Sending data...
Sending data...
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Space left in array
Array full. Size:10
����������
Sending data...
```

It seems, that the data from the module was send with a big time delay.
For my understanding the data from module was transferred with an inappropriate baud rate, resulting in weird characters.

**The following questions arise for me:**
- How do I find the correct baud rate of the RX data so as not to get strange characters?
- Which string does the module return exactly when sending a 'k' ?
   (an example would be desirable)
- Do I have to/should I send several 'k's to the module to respond?
- How to use this method to take multiple measurements per second if the response from the module is (apparently) delayed?

will-code-for-pizza, 2023-10-16:
Here is the sketch without unnessecary output to serial console.
Just to avoid UART communication issues.

```
#include <Arduino.h>
#include <SoftwareSerial.h>
#define ARRAY_SIZE 10 // Size of array to store data

char receivedData[ARRAY_SIZE]; // Array to store data into
int dataCounter = 0; // Counter for received data

bool debugMode = false;

void setup() {
  // Init serial console baudrate
  Serial.begin(57600);

  Serial.println("START");
  
  // Init UART console baudrate
  Serial1.begin(57600);
  SoftwareSerial Serial1(0, 1); // RX, TX

  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }  
}

void loop() {
  // Send string 'k' via UART
  if (debugMode == true) {
    Serial.println("Sending data...");
  }
  Serial1.println('k');

  // Waiting for UART data
  if (Serial1.available()) {
    // Check if there is space left in array to store data
    while (dataCounter < ARRAY_SIZE) {
      if (debugMode) {
        Serial.println("Space left in array");
      }
      char receivedChar = Serial1.read();
      receivedData[dataCounter] = receivedChar;
      dataCounter++;
    }

    int size = sizeof(receivedData);
    if (debugMode) {
      Serial.print("Array full. "); 
      Serial.print("Size:");
      Serial.println(size);
    }

    for (int i = 0; i < ARRAY_SIZE; i++) {
      Serial.print(receivedData[i]);
    }
    Serial.println(); // New line
    // Reset counter
    dataCounter = 0;
  }
  delay(2000);
}
```

will-code-for-pizza, 2023-10-17:
Update:

Issue with weird chars solved.  -->  Changed

`char receivedData[ARRAY_SIZE];`

to

`int receivedData[ARRAY_SIZE];`

limpkin, 2023-10-17:
one of my customers is also manually rebooting the CDM324 using the reset input (by driving it low for 100ms)... maybe you could do this too?

will-code-for-pizza, 2023-10-17:
> one of my customers is also manually rebooting the CDM324 using the reset input (by driving it low for 100ms)... maybe you could do this too?

It's not entirely clear to me at the moment how this could help me with my problems.
Oh yes, of course I often switched the module OFF and ON while the UART communication was running in the sketch. Unfortunately without success.

will-code-for-pizza, 2023-10-17:
Oh yes, I almost forgot... Can you please put the two firmwares for 57600 and 1MBit here in the GIT?

limpkin, 2023-10-27:
I've added the bin files to the release tab.

> How do I find the correct baud rate of the RX data so as not to get strange characters?  

You should set the baud rate that matches the firmware you've flashed.

> Which string does the module return exactly when sending a 'k' ?

debug_printf("%d\r\n", speed);

> Do I have to/should I send several 'k's to the module to respond?

one char per query

> How to use this method to take multiple measurements per second if the response from the module is (apparently) delayed?

the response won't be delayed

I'm very sorry but there's so much I can help you with without actually digging into your personal code.  
I've had several other customers that managed to do what you're trying to do, I'll reach out to them to see if they're willing to share code.

limpkin, 2023-10-29:
sample arduino code has been put on the readme, tested.

closed 2023-10-29 by limpkin

will-code-for-pizza, 2023-10-30:
> sample arduino code has been put on the readme, tested.

Thanks for the arduino example
https://github.com/limpkin/cdm324_fft#sample-arduino-code-to-fetch-speed-from-device
but did you realize during your tests, that

- speed is only displayed once, then 0 once and then the output stops ?

```
07:18:11.885 -> Hello there!
07:18:11.917 -> Received from the CDM324:
07:18:11.917 -> CDM324 fw v0.1, compiled Oct  3 2023 16:11:55
07:18:11.917 -> 
07:18:11.950 -> 33.90
07:18:13.975 -> 0.00
```

- script crashes in 30% of cases ?

```
07:04:47.190 -> Firmware name: "/tmp/arduino/sketches/3FFDA1E6168788B49E3D5492DF3D250C/sketch_oct30a.ino", compiled on: Oct 30 2023
07:04:47.190 -> Fault on interrupt or bare metal(no OS) environment
07:04:47.222 -> ===== Thread stack information =====
07:04:47.222 ->   addr: 20007ed8    data: 000108b5
07:04:47.222 ->   addr: 20007edc    data: 40046f00
07:04:47.255 ->   addr: 20007ee0    data: 00000000
07:04:47.255 ->   addr: 20007ee4    data: 00009e67
07:04:47.255 ->   addr: 20007ee8    data: 000108b5
07:04:47.287 ->   addr: 20007eec    data: 0000421f
07:04:47.287 ->   addr: 20007ef0    data: 000108b5
07:04:47.287 ->   addr: 20007ef4    data: 000074c3
07:04:47.287 ->   addr: 20007ef8    data: 000074b9
07:04:47.287 ->   addr: 20007efc    data: 00002599
07:04:47.287 -> ====================================
07:04:47.287 -> =================== Registers information ====================
07:04:47.319 ->   R0 : 00000042  R1 : 40070120  R2 : 00000000  R3 : 00000000
07:04:47.319 ->   R12: ffffffff  LR : 000086a1  PC : 00009e00  PSR: 60000000
07:04:47.352 -> ==============================================================
07:04:47.352 -> Usage fault is caused by attempts to switch to an invalid state (e.g., ARM)
07:04:47.385 -> Show more call stack info by run: addr2line -e "/tmp/arduino/sketches/3FFDA1E6168788B49E3D5492DF3D250C/sketch_oct30a.ino".elf -a -f 00009e00 000086a0 00009e66 0000421e 000074c2 000074b8
```
The cause could be that in .arduino15/packages/arduino/hardware/renesas_uno/1.0.4/cores/arduino/Serial.cpp:114

```
  if(init_ok) {
    tx_done = false;
    R_SCI_UART_Write(&uart_ctrl, &c, 1);
    while (!tx_done) {}   //      <<<------ !!!
    return 1;
  }
```
more than one simultaneous access to the serial port should be attempted.
On Arduino, thats not allowed/possible.

At least the one-time output of the speed now works and the documentation is now better when it comes to Arduino.

Thanks a lot and keep grinding!

limpkin, 2023-10-30:
It didn't crash for me, and you'll get 0 when no movement is detected.