limpkin/cdm324_fft #3: Issues with sample arduino code

will-code-for-pizza, 2023-10-30:
> sample arduino code has been put on the readme, tested.

Thanks for the arduino example
https://github.com/limpkin/cdm324_fft#sample-arduino-code-to-fetch-speed-from-device
but did you realize during your tests, that

- speed is only displayed once, then 0 once and then the output stops ?

```
07:18:11.885 -> Hello there!
07:18:11.917 -> Received from the CDM324:
07:18:11.917 -> CDM324 fw v0.1, compiled Oct  3 2023 16:11:55
07:18:11.917 -> 
07:18:11.950 -> 33.90
07:18:13.975 -> 0.00
```

- script crashes in 30% of cases ?

```
07:04:47.190 -> Firmware name: "/tmp/arduino/sketches/3FFDA1E6168788B49E3D5492DF3D250C/sketch_oct30a.ino", compiled on: Oct 30 2023
07:04:47.190 -> Fault on interrupt or bare metal(no OS) environment
07:04:47.222 -> ===== Thread stack information =====
07:04:47.222 ->   addr: 20007ed8    data: 000108b5
07:04:47.222 ->   addr: 20007edc    data: 40046f00
07:04:47.255 ->   addr: 20007ee0    data: 00000000
07:04:47.255 ->   addr: 20007ee4    data: 00009e67
07:04:47.255 ->   addr: 20007ee8    data: 000108b5
07:04:47.287 ->   addr: 20007eec    data: 0000421f
07:04:47.287 ->   addr: 20007ef0    data: 000108b5
07:04:47.287 ->   addr: 20007ef4    data: 000074c3
07:04:47.287 ->   addr: 20007ef8    data: 000074b9
07:04:47.287 ->   addr: 20007efc    data: 00002599
07:04:47.287 -> ====================================
07:04:47.287 -> =================== Registers information ====================
07:04:47.319 ->   R0 : 00000042  R1 : 40070120  R2 : 00000000  R3 : 00000000
07:04:47.319 ->   R12: ffffffff  LR : 000086a1  PC : 00009e00  PSR: 60000000
07:04:47.352 -> ==============================================================
07:04:47.352 -> Usage fault is caused by attempts to switch to an invalid state (e.g., ARM)
07:04:47.385 -> Show more call stack info by run: addr2line -e "/tmp/arduino/sketches/3FFDA1E6168788B49E3D5492DF3D250C/sketch_oct30a.ino".elf -a -f 00009e00 000086a0 00009e66 0000421e 000074c2 000074b8
```
The cause could be that in .arduino15/packages/arduino/hardware/renesas_uno/1.0.4/cores/arduino/Serial.cpp:114

```
  if(init_ok) {
    tx_done = false;
    R_SCI_UART_Write(&uart_ctrl, &c, 1);
    while (!tx_done) {}   //      <<<------ !!!
    return 1;
  }
```
more than one simultaneous access to the serial port should be attempted.
On Arduino, thats not allowed/possible.

At least the one-time output of the speed now works and the documentation is now better when it comes to Arduino.

Thanks a lot and keep grinding!

_Originally posted by @will-code-for-pizza in https://github.com/limpkin/cdm324_fft/issues/2#issuecomment-1784561970_

limpkin, 2023-10-30:
it does not crash for me at all...
As you have one setup that crashes, perhaps you can contribute to this repository by fixing the issue and issuing a pull request?

will-code-for-pizza, 2023-10-30:
> it does not crash for me at all... As you have one setup that crashes, perhaps you can contribute to this repository by fixing the issue and issuing a pull request?

Of course, I will provide you with a fix, if I have found one.

**First, lets compare hardware and cabling:**

My setup:  Arduino UNO R4 Wifi, CDM324_fft module with LCD and 57600 baud firmware

My cabling:
```
CMD324                  <-->               < -->                 Arduino UNO R4
=============================================================================
nRESET                                                                  PIN 2 digital
5V_IN                   external 5V HW-131 5V OUT
GND                     external 5V HW-131 GND
MCU UART_RX                                                      PIN 1 (TX)
MCU UART_TX                                                      PIN 0 (RX)
```

limpkin, 2023-10-30:
great, looking forward to it!

fluppie, 2023-11-19:
I see the same, flashed the 57600 baud version, send "k" to the terminal (Arduino IDE) and I get one time the speed.
If I press the reset button, I can do it again.

Maybe a continuous sending of the speed to the terminal can also be nice? don't know how many times per second it can send the speed?

limpkin, 2023-12-05:
how's your progress @will-code-for-pizza ?

closed 2024-02-23 by limpkin