limpkin/cdm324_fft #4: Trouble installing pyqt5 (Win 10 22H2 with Python 3.7.3)

fluppie, 2023-11-16:
Hi, bought the kit via Lectronz (4158) Any idea why the building is failing? What Python version are you using?
I have python 3.7.3 and pip 23.3 on W10 22H2

```
PS C:\dopplerradar\scripts> pip install pyqt5 chaco stm32loader
Collecting pyqt5
  Using cached PyQt5-5.15.10-cp37-abi3-win_amd64.whl.metadata (2.2 kB)
Requirement already satisfied: chaco in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (6.0.0)
Requirement already satisfied: stm32loader in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (0.5.1)
Collecting PyQt5-sip<13,>=12.13 (from pyqt5)
  Using cached PyQt5_sip-12.13.0.tar.gz (123 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyQt5-Qt5>=5.15.2 (from pyqt5)
  Using cached PyQt5_Qt5-5.15.2-py3-none-win_amd64.whl (50.1 MB)
Requirement already satisfied: traits>=6.2.0 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from chaco) (6.4.3)
Requirement already satisfied: traitsui in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from chaco) (8.0.0)
Requirement already satisfied: pyface>=7.2.0 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from chaco) (8.0.0)
Requirement already satisfied: numpy in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from chaco) (1.18.2)
Requirement already satisfied: enable>=5.2.0 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from enable[layout,svg]>=5.2.0->chaco) (6.0.0)
Requirement already satisfied: pyserial in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from stm32loader) (3.4)
Requirement already satisfied: progress in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from stm32loader) (1.6)
Requirement already satisfied: pillow in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from enable>=5.2.0->enable[layout,svg]>=5.2.0->chaco) (9.5.0)
Requirement already satisfied: fonttools in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from enable>=5.2.0->enable[layout,svg]>=5.2.0->chaco) (4.38.0)
Requirement already satisfied: kiwisolver in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from enable[layout,svg]>=5.2.0->chaco) (1.4.5)
Requirement already satisfied: pyparsing in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from enable[layout,svg]>=5.2.0->chaco) (2.4.6)
Requirement already satisfied: importlib-metadata>=3.6 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from pyface>=7.2.0->chaco) (6.7.0)
Requirement already satisfied: importlib-resources>=1.1.0 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from pyface>=7.2.0->chaco) (5.12.0)
Requirement already satisfied: zipp>=0.5 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from importlib-metadata>=3.6->pyface>=7.2.0->chaco) (3.1.0)
Requirement already satisfied: typing-extensions>=3.6.4 in c:\users\philippe\appdata\local\programs\python\python37\lib\site-packages (from importlib-metadata>=3.6->pyface>=7.2.0->chaco) (4.7.1)
Using cached PyQt5-5.15.10-cp37-abi3-win_amd64.whl (6.8 MB)
Building wheels for collected packages: PyQt5-sip
  Building wheel for PyQt5-sip (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for PyQt5-sip (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [42 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'PyQt5.sip' extension
      creating build
      creating build\temp.win-amd64-cpython-37
      creating build\temp.win-amd64-cpython-37\Release
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" /Tcapiversions.c /Fobuild\temp.win-amd64-cpython-37\Release\apiversions.obj
      apiversions.c
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" /EHsc /Tpbool.cpp /Fobuild\temp.win-amd64-cpython-37\Release\bool.obj
      bool.cpp
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" /EHsc /Tcdescriptors.c /Fobuild\temp.win-amd64-cpython-37\Release\descriptors.obj
      descriptors.c
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" /EHsc /Tcint_convertors.c /Fobuild\temp.win-amd64-cpython-37\Release\int_convertors.obj
      int_convertors.c
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt" /EHsc /Tcobjmap.c /Fobuild\temp.win-amd64-cpython-37\Release\objmap.obj
      objmap.c
      objmap.c(99): warning C4311: 'type cast': pointer truncation from 'void *' to 'unsigned long'
      objmap.c(100): warning C4311: 'type cast': pointer truncation from 'void *' to 'unsigned long'
      "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\include -IC:\Users\Philippe\AppData\Local\Programs\Python\Python37\Include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Win [... cut ...]
      LINK : fatal error LNK1158: cannot run 'rc.exe'
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit code 1158
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for PyQt5-sip
Failed to build PyQt5-sip
ERROR: Could not build wheels for PyQt5-sip, which is required to install pyproject.toml-based projects
WARNING: There was an error checking the latest version of pip.
```

limpkin, 2023-11-16:
Hello!
Did you correctly install the C++ build tools? 
see https://github.com/limpkin/cdm324_fft/tree/main/scripts

limpkin, 2023-11-16:
can you remove the VS2015?

fluppie, 2023-11-16:
One step closer, it build, but running _start.py does not work :).

```
PS C:\dopplerradar\scripts> python _start.py
Traceback (most recent call last):
  File "_start.py", line 244, in <module>
    app = CDM324MonitorApp()
  File "_start.py", line 237, in __init__
    self.com_port_monitor()
  File "_start.py", line 134, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
  File "C:\dopplerradar\scripts\cdm324_device.py", line 29, in connect
    self.serial = serial.Serial(self.com_port, 1000000)
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\site-packages\serial\serialwin32.py", line 78, in open
    self._reconfigure_port()
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'De parameter is onjuist.', None, 87)
PS C:\dopplerradar\scripts>
```

The device is assigned COM2 in the device manager. I have the complete kit, with USB isolator and expansion board.

limpkin, 2023-11-16:
how many COM ports do you have in the device manager? can you share a screen capture?

fluppie, 2023-11-16:
[image]

And the CH340 COM2 is from your board

limpkin, 2023-11-16:
that's a fun one... you're correctly selecting COM2 in the GUI I assume? 
Maybe update your python install? :)

fluppie, 2023-11-16:
What version are you running 3.12?
To start the scripts, I tried running all 3 py files. Anything I missed?
Is "python _start.py" the way to go? 

```
PS C:\dopplerradar\scripts> python stream_visualizer.py
Traceback (most recent call last):
  File "stream_visualizer.py", line 269, in <module>
    s = serial.Serial(port)
AttributeError: module 'serial' has no attribute 'Serial'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "stream_visualizer.py", line 272, in <module>
    except (OSError, serial.SerialException):
AttributeError: module 'serial' has no attribute 'SerialException'
PS C:\dopplerradar\scripts> python cdm324_device.py
PS C:\dopplerradar\scripts> python _start.py
Traceback (most recent call last):
  File "_start.py", line 31, in scan_available_com_ports
    s = serial.Serial(port)
AttributeError: module 'serial' has no attribute 'Serial'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "_start.py", line 244, in <module>
    app = CDM324MonitorApp()
  File "_start.py", line 237, in __init__
    self.com_port_monitor()
  File "_start.py", line 118, in com_port_monitor
    self.current_com_ports = self.scan_available_com_ports()
  File "_start.py", line 34, in scan_available_com_ports
    except (OSError, serial.SerialException):
AttributeError: module 'serial' has no attribute 'SerialException'
```

limpkin, 2023-11-17:
3.12 should work indeed. 
python _start.py is indeed the way to go
I think you have the wrong serial module installed.
can you do:
- pip uninstall serial
- pip install pyserial

fluppie, 2023-11-17:
I'm uninstalling all old Python versions and will re-install 3.12 and hopefully that will work. I had 5 different versions installed over the years.

fluppie, 2023-11-17:
still struggling. Could the reason by that I cannot select 1 000 000 baud from the dropdown? Cannot go above 128 000 baud.
[image]

fluppie, 2023-11-17:
With Python 3.12 I'm unable to install/compile chaco. With Python 3.11 chaco installs/compiles fine.
After installing 311 I tried to "python _start.py"

Had to "pip install scipy" and "pip install keyboard"

Then I get again this error:
```

PS C:\dopplerradar\scripts> python _start.py
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 244, in <module>
    app = CDM324MonitorApp()
          ^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 237, in __init__
    self.com_port_monitor()
  File "C:\dopplerradar\scripts\_start.py", line 134, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\cdm324_device.py", line 29, in connect
    self.serial = serial.Serial(self.com_port, 1000000)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'De parameter is onjuist.', None, 87)
PS C:\dopplerradar\scripts>
```

limpkin, 2023-11-17:
did you 
-    pip uninstall pyserial
-    pip install serial 
?

fluppie, 2023-11-17:
yes

```
PS C:\dopplerradar\scripts> pip uninstall pyserial
WARNING: Skipping pyserial as it is not installed.
PS C:\dopplerradar\scripts> pip uninstall serial
WARNING: Skipping serial as it is not installed.
PS C:\dopplerradar\scripts> pip install serial
Collecting serial
  Using cached serial-0.0.97-py2.py3-none-any.whl (40 kB)
Requirement already satisfied: future>=0.17.1 in c:\python311\lib\site-packages (from serial) (0.18.3)
Requirement already satisfied: pyyaml>=3.13 in c:\python311\lib\site-packages (from serial) (6.0.1)
Requirement already satisfied: iso8601>=0.1.12 in c:\python311\lib\site-packages (from serial) (2.1.0)
Installing collected packages: serial
Successfully installed serial-0.0.97
PS C:\dopplerradar\scripts> python _start.py
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 31, in scan_available_com_ports
    s = serial.Serial(port)
        ^^^^^^^^^^^^^
AttributeError: module 'serial' has no attribute 'Serial'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 244, in <module>
    app = CDM324MonitorApp()
          ^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 237, in __init__
    self.com_port_monitor()
  File "C:\dopplerradar\scripts\_start.py", line 118, in com_port_monitor
    self.current_com_ports = self.scan_available_com_ports()
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 34, in scan_available_com_ports
    except (OSError, serial.SerialException):
                     ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'serial' has no attribute 'SerialException'
PS C:\dopplerradar\scripts>
```

limpkin, 2023-11-17:
sorry i made a mistake.... could you instead:
- pip uninstall serial
- pip install pyserial

fluppie, 2023-11-18:
I already tried that :), I saw an earlier post of you was edited.
Do you think it has to do with the 1 Mbit / 1 000 000 baud speed?
Could you make the script we can do something like: "python _start.py -p COM2" ? Forcing the script to use the right/specific COM port.

```
PS C:\dopplerradar\scripts> pip uninstall serial
Found existing installation: serial 0.0.97
Uninstalling serial-0.0.97:
  Would remove:
    c:\python311\lib\site-packages\serial-0.0.97.dist-info\*
    c:\python311\lib\site-packages\serial\*
Proceed (Y/n)? y
  Successfully uninstalled serial-0.0.97
PS C:\dopplerradar\scripts> pip uninstall pyserial
WARNING: Skipping pyserial as it is not installed.
PS C:\dopplerradar\scripts> pip install pyserial
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Installing collected packages: pyserial
Successfully installed pyserial-3.5
PS C:\dopplerradar\scripts> python _start.py
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 244, in <module>
    app = CDM324MonitorApp()
          ^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 237, in __init__
    self.com_port_monitor()
  File "C:\dopplerradar\scripts\_start.py", line 134, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\cdm324_device.py", line 29, in connect
    self.serial = serial.Serial(self.com_port, 1000000)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'De parameter is onjuist.', None, 87)
PS C:\dopplerradar\scripts>
```

limpkin, 2023-11-18:
well it does look like the program tries to open the port... it looks like something else now.
maybe try these suggestions? https://stackoverflow.com/questions/20879939/pyserial-error-cannot-open-port

fluppie, 2023-11-18:
This simple code does not return any errors
```

import serial

serialPort = serial.Serial(port = "COM2", baudrate=1000000,
                           bytesize=8, timeout=2, stopbits=serial.STOPBITS_ONE)

serialPort.close()
```

Testing the code on the other COM ports in my system does return some errors now and then.
```

PS C:\dopplerradar\scripts> py comtestcloseCOM2.py
PS C:\dopplerradar\scripts> py comtestcloseCOM3.py
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\comtestcloseCOM3.py", line 3, in <module>
    serialPort = serial.Serial(port = "COM3", baudrate=1000000,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'De parameter is onjuist.', None, 87)
PS C:\dopplerradar\scripts> py comtestcloseCOM4.py
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\comtestcloseCOM4.py", line 3, in <module>
    serialPort = serial.Serial(port = "COM4", baudrate=1000000,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM4': OSError(22, 'De time-outperiode van de semafoor is verlopen.', None, 121)
PS C:\dopplerradar\scripts> py comtestcloseCOM5.py
PS C:\dopplerradar\scripts>
```

I then tried to change line 25 in _start.py to:
`		ports = ['COM%s' % (i + 1) for i in range(7,256)]`
So the first COM port it can find is COM8, changed the CH340 from COM2 to COM8 and voila, tiny step further:
[image]

Now we need to know why it cannot open the com port. Is there a way to force the script to use COM8 and not do any com port scan?

limpkin, 2023-11-18:
I'm afraid not, as your error happens when python tries to talk with COM2: https://github.com/limpkin/cdm324_fft/blob/main/scripts/_start.py#L133
at this line, could you add a print(com_port) ? it should report COM2.
worst case, could you add here: https://github.com/limpkin/cdm324_fft/blob/main/scripts/_start.py#L133 the following line: return ["COM2"]

fluppie, 2023-11-19:
Because of line 25 https://github.com/limpkin/cdm324_fft/blob/main/scripts/_start.py#L25
ports = ['COM%s' % (i + 1) for i in range(2,256)]
COM3 is the earliest com port that can be found, so that's why I changed the com port in the W10 device manager from com2 to com 8 and changed the detection range to (5,256). 

Now it finds COM8 (this is correct, the cdm324 device) and that gives me this:
[image]

limpkin, 2023-11-19:
no button needs to be pressed indeed.
just to make sure the serial link is working, could you try the reflashing procedure ( https://github.com/limpkin/cdm324_fft#reflashing-the-device ) with https://github.com/limpkin/cdm324_fft/releases/download/v1.00/cdm324_1000000.bin ?

fluppie, 2023-11-19:
I can also make the debug tool work when hardcoding the com port
```
		# Launch debug tool
		self.com_port = "COM8"
		stream_vis_start(self.com_port)
```

[image]

But I need to reset the device to re-run the script after closing the window. Probably because the com_port isn't closed properly with this hardcoded COM8.
```
PS C:\dopplerradar\scripts> python _start.py
COM8
2023-11-19 21:10:17.086388: skipping one frame
2023-11-19 21:10:26.237943: skipping one frame
2023-11-19 21:10:26.407199: skipping one frame
2023-11-19 21:10:26.718888: skipping one frame
2023-11-19 21:10:28.005532: skipping one frame
2023-11-19 21:10:28.637009: skipping one frame
2023-11-19 21:10:30.348279: skipping one frame
2023-11-19 21:10:31.979112: skipping one frame
2023-11-19 21:10:32.059897: skipping one frame
2023-11-19 21:10:34.920247: skipping one frame
2023-11-19 21:10:37.286116: skipping one frame
2023-11-19 21:10:38.058548: skipping one frame
2023-11-19 21:10:38.266316: skipping one frame
2023-11-19 21:10:38.742160: skipping one frame
2023-11-19 21:10:38.973134: skipping one frame
2023-11-19 21:10:39.115380: skipping one frame
2023-11-19 21:10:40.830078: skipping one frame
2023-11-19 21:10:41.542726: skipping one frame
2023-11-19 21:10:41.805361: skipping one frame
COM8
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python311\Lib\tkinter\__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\tkinter\__init__.py", line 861, in callit
    func(*args)
  File "C:\dopplerradar\scripts\_start.py", line 137, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\cdm324_device.py", line 44, in connect
    self.cdm324_version = self.cdm324_version[i:-2].decode('utf-8')
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x81 in position 8: invalid start byte
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 249, in <module>
    app.mainloop()
  File "C:\Python311\Lib\tkinter\__init__.py", line 1485, in mainloop
    self.tk.mainloop(n)
  File "C:\Python311\Lib\tkinter\__init__.py", line 1943, in __call__
    def __call__(self, *args):

KeyboardInterrupt
PS C:\dopplerradar\scripts> python _start.py
COM8
Traceback (most recent call last):
  File "C:\dopplerradar\scripts\_start.py", line 248, in <module>
    app = CDM324MonitorApp()
          ^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 241, in __init__
    self.com_port_monitor()
  File "C:\dopplerradar\scripts\_start.py", line 137, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\cdm324_device.py", line 44, in connect
    self.cdm324_version = self.cdm324_version[i:-2].decode('utf-8')
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 1-2: invalid continuation byte
PS C:\dopplerradar\scripts> python _start.py
COM8
```

limpkin, 2023-11-19:
just to make sure we're on the same page:
if you modify https://github.com/limpkin/cdm324_fft/blob/main/scripts/_start.py#L133 to add the following line: return ["COM8"] you do get the "No COM ports detected" message?

fluppie, 2023-11-19:
Yes, see video :).

https://github.com/limpkin/cdm324_fft/assets/8402106/3962455d-9ed1-4066-b95d-8ccc725d4c0f

limpkin, 2023-11-19:
something odd is happening... the "no com ports detected!" only gets displayed if scan_available_com_ports return no com ports ( scan_available_com_ports ).
However, you modified scan_available_com_ports to return one com port.... 
maybe change self.current_com_ports = self.scan_available_com_ports()to self.current_com_ports = ["COM8"] ?

fluppie, 2023-11-19:
Then it doesn't work anymore, sadly :)

```
PS C:\dopplerradar\scripts> python _start.py
COM8
COM8
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python311\Lib\tkinter\__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\tkinter\__init__.py", line 861, in callit
    func(*args)
  File "C:\dopplerradar\scripts\_start.py", line 138, in com_port_monitor
    (self.connected_to_cdm, cdm_version) = self.device.connect(com_port)
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\cdm324_device.py", line 29, in connect
    self.serial = serial.Serial(self.com_port, 1000000)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM8': PermissionError(13, 'Toegang geweigerd.', None, 5)
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Python311\Lib\tkinter\__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 231, in <lambda>
    self.update_fw_button = tk.Button(self, text="Start Debug Tool", font=tkfont.Font(family='Helvetica', size=9), width="18", command=lambda:[self.start_debug_tool()])
                                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\dopplerradar\scripts\_start.py", line 100, in start_debug_tool
    stream_vis_start(self.com_port)
  File "C:\dopplerradar\scripts\stream_visualizer.py", line 253, in stream_vis_start
    platform_serial = serial.Serial(com_port, 1000000)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python311\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '': FileNotFoundError(2, 'Het systeem kan het opgegeven pad niet vinden.', None, 3)
PS C:\dopplerradar\scripts>
```

limpkin, 2023-11-20:
"access denied".... any reason why that would be? any software running in the background?

fluppie, 2023-11-20:
maybe the script is opening it and not closing it somewhere and then trying to open but gets an error trying to open an already open port? I'll see if I can run it on my work laptop, but it's pretty closed by IT. So I don't think I can install Python on it.

limpkin, 2023-11-20:
any other tool running on your system maybe?

fluppie, 2023-11-20:
I got the same on my worklaptop, only the com port of the module, no other com ports on that machine. Script without any modifications opens, but gives red combo and no com ports detected. Clicking the debug tools gives some similar permission error.

limpkin, 2023-12-05:
@fluppie ?

closed 2024-02-23 by limpkin

DutchMikeDutch, 2024-02-24:
@fluppie any update's i have been able to come to the same point, meaning: i am able to open streamvisualizer, nevertheless the debug tool trows errors.

If i use Arduino Serial Monitor: on my com port i am able to see data passing.

Also have not been able to use my esp8266 or esp8266 esphome to read the uart to be able to display this on my 7 segment display. the default baud_rate 1000000 is also causing me challenges. python is definity not my strong suit hence i am severly blocked.

Using windows 11 btw, hence keep in mind you would require to remove some blockings like downloaded files