limpkin/cdm324_fft #5: dopplersensor display stuck at 0.4

Marcus-Peterson, 2024-02-08:
So, I have a voltage divider the delivers electricity to the dopplersensor, whenever I connect the 5v and GND to the mb102, the display shifts between a bunch of numbers (if I wave my hand in front of it for example). And after a couple of seconds it's stuck at 0.4, and sometimes all I can see is a segment go around in circles. 

I've tested to see if there was something wrong with the doppler sensor through the expansion board and connected it via usbc to my pc, but nothing. It seems fine..? It detects speeds flawlessly. 

When a/the segment goes around in circles it is incapable of detecting movements

What is going on here?

limpkin, 2024-02-08:
hello,
Can you expand on " I have a voltage divider the delivers electricity to the dopplersensor"?
Is that the mb102 you're referring to? https://www.az-delivery.de/en/products/mb102-breadboard
could you send a picture of your wiring diagram?

will-code-for-pizza, 2024-03-18:
I have a similar issue:

If I connect the CDM324 "standalone" via expansion board to USB 5V, measuring speed works fine.
Module detects movements and if nothing happened, the display shows circling line.

If I connect CDM324 to Arduino

```
CMD324   <----->   Arduino UNO R4 Wifi
=================================
nRESET             PIN 2 digital
5V_IN              5V (do NOT use external power supply or script will crash!)
GND                GND
MCU UART_RX        PIN 1 (TX)
MCU UART_TX        PIN 0 (RX)
```

display shows a constant speed ~ 4 km/h (range from 3.80 to 4.20 km/h)

will-code-for-pizza, 2024-03-18:
Additional video from issue:

https://youtube.com/shorts/JUVHfLQQhDU

limpkin, 2024-03-18:
@will-code-for-pizza your issue seems to be related to a noisy power supply, possibly injecting noise into the CDM324

will-code-for-pizza, 2024-03-19:
> @will-code-for-pizza your issue seems to be related to a noisy power supply, possibly injecting noise into the CDM324

I shouldn't have bought that Chinese junk... (the power supply)    :-(

closed 2024-03-25 by Marcus-Peterson