limpkin/cdm324_fft #8: Changing radius off CDM324

Marcus-Peterson, 2024-04-08:
Is it possible to programmatically change the circumference of the radius of the CDM324?

limpkin, 2024-04-08:
do you mean its radiation pattern? that's something that is defined by its radiating elements

Marcus-Peterson, 2024-04-09:
No, the detection radius of speed

limpkin, 2024-04-09:
what do you mean by "radius"? do you mean "range"?

Marcus-Peterson, 2024-04-09:
Well, the range of the actual cdm324 is some value, however. Detection is not in a straight line, that is why is was curious about the actual radius of the detection range. If I'm thinking about this correctly.

Marcus-Peterson, 2024-04-15:
Yeah, I guess so. I have no knowledge regarding the math. 
But I was thinking akin to this:
[image]

limpkin, 2024-04-15:
the drawing you made is of the radiation pattern, which can't be programmatically changed as it directly depends on the antenna design.

closed 2024-04-15 by limpkin