limpkin/cdm324_fft #9: CDM324 Monitoring Tool does not connect

henridbr, 2024-04-22:
Hi,

Using the bootloader I was able to change the baud rate to 57600.

At first I replaced the baud rate of 1000000 by 57600 in the scripts: 

-  stream_visualizer.py (lines 253-254) 
-  cdm324_device.py (lines 29-30). 

I connected the expansion board to my computer by USB, it's on COM8 set at 57600 bauds. 

Then : `python "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py" `
and I got the CDM324 Monitoring Tool. 

Unfortunately it did not connect. The port field changed alternatively from COM4 to COM8 with the background color in white, but nothing occurs. 
When I disconnected the USB cable, the backcolor of the COM field became red with the warning "No COM ports detected!" 

Pushing the Debug button, I got some information: 

> PS C:\Users\henri> python "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py"
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.1008.0_x64__qbz5n2kfra8p0\Lib\tkinter\__init__.py", line 1967, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py", line 226, in <lambda>
    self.update_fw_button = tk.Button(self, text="Start Debug Tool", font=tkfont.Font(family='Helvetica', size=9), width="18", command=lambda:[self.start_debug_tool()])
                                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py", line 98, in start_debug_tool
    stream_vis_start(self.com_port)
  File "C:\Users\henri\...\cdm324_fft-main\scripts\stream_visualizer.py", line 253, in stream_vis_start
    platform_serial = serial.Serial(com_port, 57600)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '': FileNotFoundError(2, 'Le chemin d’accès spécifié est introuvable.', None, 3)
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.1008.0_x64__qbz5n2kfra8p0\Lib\tkinter\__init__.py", line 1967, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py", line 226, in <lambda>
    self.update_fw_button = tk.Button(self, text="Start Debug Tool", font=tkfont.Font(family='Helvetica', size=9), width="18", command=lambda:[self.start_debug_tool()])
                                                                                                                                               ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henri\...\cdm324_fft-main\scripts\_start.py", line 98, in start_debug_tool
    stream_vis_start(self.com_port)
  File "C:\Users\henri\...\cdm324_fft-main\scripts\stream_visualizer.py", line 253, in stream_vis_start
    platform_serial = serial.Serial(com_port, 57600)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\henri\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '': FileNotFoundError(2, 'Le chemin d’accès spécifié est introuvable.', None, 3)
>

Marcus-Peterson, 2024-04-23:
Could you share your code here so that we can help?

limpkin, 2024-04-23:
the rx_size & tx_size value shouldn't be changed (they are buffer size values). Only the serial.Serial() value should be changed.
Can you confirm that a standard terminal program allows you to talk to the platform?

henridbr, 2024-04-24:
I tried both buffer sizes, same result with the GUI and debug. 
Connected by USB to the expansion board with the port set at 57600 bauds, I receive the boot message (reset button) and some speed values but not each time I send "k".
It works fine with the FTDI232.

limpkin, 2024-04-24:
So I just re-tried on my end flashing the 57600 firmware (with the _start.py script), modifying line 29 of _start.py to change '1000000' with '57600' and managed to get the GUI (not debug tool) running.
Please note that the debug tool will NOT work with a 57600 firmware as the baud rate will simply limit what's possible to broadcast.